lxi-tools/liblxi #8: ./configure doesn't handle /usr/include/tirpc/rpc/rpc.h

robert-scheck, 2018-03-17:
As per https://fedoraproject.org/wiki/Changes/SunRPCRemoval, the `/usr/include/rpc/rpc.h` is moved to `/usr/include/tirpc/rpc/rpc.h` (upstream change) which breaks liblxi `./configure`, because it does not detect this file.

lundmar, 2018-03-17:
Thanks for the info.

I've just pushed 69871f0334690d34695e0dd320bee09896a6cf41 which should fix it.

robert-scheck, 2018-03-21:
Not really. Using libtirpc-1.0.3-0.fc29 and rpcgen-1.3.1-3.fc28 (Rawhide), I end up like this:

```
+ /usr/bin/make -O -j48
Making all in src
make[1]: Entering directory '/builddir/build/BUILD/liblxi-1.12/src'
rpcgen -M vxi11core.rpcl
make[1]: Leaving directory '/builddir/build/BUILD/liblxi-1.12/src'
This implementation doesn't support newstyle or MT-safe code!
usage: rpcgen infile
	rpcgen [-abkCLNTM][-Dname[=value]] [-i size] [-I [-K seconds]] [-Y path] infile
	rpcgen [-c | -h | -l | -m | -t | -Sc | -Ss | -Sm] [-o outfile] [infile]
	rpcgen [-s nettype]* [-o outfile] [infile]
	rpcgen [-n netid]* [-o outfile] [infile]
options:
-a		generate all files, including samples
-b		backward compatibility mode (generates code for SunOS 4.1)
-c		generate XDR routines
-C		ANSI C mode
-Dname[=value]	define a symbol (same as #define)
-h		generate header file
-i size		size at which to start generating inline code
-I		generate code for inetd support in server (for SunOS 4.1)
-K seconds	server exits after K seconds of inactivity
-l		generate client side stubs
-L		server errors will be printed to syslog
-m		generate server side stubs
-M		generate MT-safe code
-n netid	generate server code that supports named netid
-N		supports multiple arguments and call-by-value
-o outfile	name of the output file
-s nettype	generate server code that supports named nettype
-Sc		generate sample client code that uses remote procedures
-Ss		generate sample server code that defines remote procedures
-Sm 		generate makefile template 
-t		generate RPC dispatch table
-T		generate code to support RPC dispatch tables
-Y path		directory name to find C preprocessor (cpp)
-5		SysVr4 compatibility mode
--help		give this help list
--version	print program version
make[1]: *** [Makefile:790: vxi11core_clnt.c] Error 1
```

dmgk, 2018-03-21:
https://github.com/lxi-tools/liblxi/commit/69871f0334690d34695e0dd320bee09896a6cf41 breaks FreeBSD build (there's no pkgconfig for Sun RPC, it's in base).

lundmar, 2018-03-21:
It's not easy to make checks that work on different platforms and different versions of same platforms.

@dmgk You will have to use libtirpc_CFLAGS and libtirpc_LIBS to override the pkg-config checks.

robert-scheck, 2018-03-21:
Wouldn't it make sense to have something like (metacode):

```
if !AC_CHECK_HEADERS(rpc/rpc.h) then
  PKG_CHECK_MODULES([libtirpc], [libtirpc])
fi
```

lundmar, 2018-03-21:
Something like that could be done yes.

robert-scheck, 2018-03-24:
That would be great, because it would also cover other plattforms properly.

lundmar, 2018-03-24:
Ok. Fixed in 1233cb11b62f72dd2e5ab2364a9eba7b03140140. It's not elegant but it gets the job done.

closed 2018-03-24 by lundmar