lxi-tools/liblxi #11: Question about liblxi usage

AlteredWildebeest, 2018-08-12:
Hello,
I've compiled and built the liblxi and lxi-tools, and they both are working. Now I'm trying to write my own code using the library. The first thing I did was try to compile the liblxi/test/scpi.c file. Whenever I attempt compilation I get undefined reference errors:
/tmp/ccIEtpsP.o: In function `main':
scpi.c:(.text+0x30): undefined reference to `lxi_init'
scpi.c:(.text+0x52): undefined reference to `lxi_connect'
scpi.c:(.text+0x87): undefined reference to `lxi_send'
scpi.c:(.text+0xa8): undefined reference to `lxi_receive'
scpi.c:(.text+0xc4): undefined reference to `lxi_disconnect'
collect2: error: ld returned 1 exit status

Any ideas on what I'm doing wrong?

Also, how does one run the GUI?

lundmar, 2018-08-12:
This type of error is typical if you forget to link with the liblxi library. You have to take care to tell your compiler that you want to link with liblxi.so by using the "-l" option, e.g. "-llxi". You also have to make sure that the library is in your compilers library search path.

The name of the GUI binary is lxi-gui.

AlteredWildebeest, 2018-08-12:
Awesome thanks very much lundmar! That fixed my linking issue. I'm new to this.

Where is lxi-gui located? The command is not available in bash. I looked through the source in lxi-tools and found the lxi-gui folder, but no compiled binary. Did a search in bash for "lxi-gui" and didnt find anything. Am I doing something wrong? Does the gui need to be compiled separately? The command "lxi" is available in bash, just not lxi-gui.

Thanks so much!
P.S. I have a lot of Siglent instruments: SDS1204, SDG2042, SDM3055, SDM3065, SPD3303X.
If there's anything you would like me to test on any of these let me know. Cheers

lundmar, 2018-08-12:
If you are compiling and installing manually you need to use the --enable-lxi-gui configure option. It will tell you which dependencies you are missing if any.

If you are new to autotools I simply recommend you install and use lxi-tools via snap:
```$ snap install lxi-tools```

closed 2018-08-22 by lundmar