lxi-tools/liblxi #12: python bindings?

aewallin, 2018-09-13:
Have you considered python bindings? Would be very handy IMO and maybe a good alternative to what I am currently using: vxi-11
https://github.com/python-ivi/python-vxi11

I could give it a try using the boost-python library if/when I have some free time!

lundmar, 2018-09-13:
The point of liblxi is to provide a simple, high performance library that can easily link with C/C++ applications without any additional dependencies such as Python.

That being said I welcome any effort trying to add Python bindings for liblxi but it will only happen if someone steps up and helps maintain it ;)

In other words, you are welcome to join the project and start working on some Python bindings - I can create a github repo for you where you can play around and release the bindings.

On the technical side, I would not look to boost-python but rather the dl module of cpython. You don't want to add a dependency to Boost when it is not strictly needed.

lundmar, 2018-09-21:
@aewallin You want to give it a try at some point? Any thoughts? :)

aewallin, 2018-09-21:
thanks for the reminder! ;)
It would be good to try out if it's straightforward or not...
The ones I did before used cmake for the build and boost-python for the wrapper - so I would use that now also - I don't mind adding these as dependencies for the py-lib but someone else might disagree.

lundmar, 2018-09-21:
Feel free to do whatever you want - I won't stop you ;)

You can start by doing some prototyping and then refine the solution later on - prototyping is the fun part :)

My only recommendation is to keep the end solution as simple as possible :smile:

lundmar, 2018-09-21:
I've created https://github.com/lxi-tools/python-liblxi and put an invite for you to access it.

Consider it a playground repo for your python stuff whenever you feel like sharing of putting something out there.

aewallin, 2018-09-24:
711f5e5  in python-liblxi is now a quick hack, trying to get liblxi to build with cmake, and a quick-and-dirty c++ boost-python wrapper in the same style I wrote for some earlier projects.
not tested at all yet - will try to test at work or at home in the next few days...

lundmar, 2018-09-24:
Looks like a good start to me :+1:

Oh, by the way, you might want to add an open source license to your code. I suggest you continue your work in the same license used by liblxi, that is, BSD-3.

lundmar, 2018-09-25:
I'm closing this issue since we now have a separate issue tracker for python-liblxi.

closed 2018-09-25 by lundmar