lxi-tools/liblxi #13: build from source instructions?

aewallin, 2018-09-21:
Could the readme please include build-from-source (e.g. git-checkout) instructions?

for example I get

> $ autoconf
> $ automake
> configure.ac:6: error: required file './ar-lib' not found
> configure.ac:6:   'automake --add-missing' can install 'ar-lib'
> configure.ac:6: error: required file './compile' not found
> configure.ac:6:   'automake --add-missing' can install 'compile'
> configure.ac:9: error: required file './config.guess' not found
> configure.ac:9:   'automake --add-missing' can install 'config.guess'
> configure.ac:9: error: required file './config.sub' not found
> configure.ac:9:   'automake --add-missing' can install 'config.sub'
> configure.ac:4: error: required file './install-sh' not found
> configure.ac:4:   'automake --add-missing' can install 'install-sh'
> configure.ac:9: error: required file './ltmain.sh' not found
> configure.ac:4: error: required file './missing' not found
> configure.ac:4:   'automake --add-missing' can install 'missing'
> configure.ac:3: error: required file 'src/include/config.h.in' not found
> src/Makefile.am: error: required file './depcomp' not found
> src/Makefile.am:   'automake --add-missing' can install 'depcomp'

and
> $ ./configure 
> configure: error: cannot find install-sh, install.sh, or shtool in "." "./.." "./../.."

some error-message suggested running
> $ aclocal
that runs without errors but doesn't help with the above.

lundmar, 2018-09-21:
The README only describes how to install from source tarball - this is the normal way of doing things.

However, I guess it is kind of a hidden fact, but if you checkout code directly from git you should run the autogen.sh script to generate the configure script etc.. This is how most autotools based projects do it.

I don't think the README should include details on how to do development nor specific examples on how to use the configure script because those details are often specific to the host system and how you prefer to install things. There have been discussions here https://github.com/lxi-tools/lxi-tools/issues/16 with @HoracioDos to put that kind of details in a wiki instead of in the README.

aewallin, 2018-09-22:
about three lines more in the README would not hurt IMO:
requiremenets: libxml2, avahi, etc.
build: autogen, configure, make, make install

lundmar, 2018-09-22:
Ok, I'll find a place to squeeze it in :)

lundmar, 2018-09-22:
Fixed in 0409f4a71164c93b972fe56d06b01c1eab147349

closed 2018-09-24 by lundmar