lxi-tools/liblxi #14: Fresh build on OSX complains about avahi-client being missing

chevdor, 2018-10-20:
The doc however mentions that avahi is optional.

soundstorm, 2018-11-26:
However you can install `rpcgen-mt` via macports, which will need an replacement of `rpcgen` in the Makefile.am
@lundmar a flag or switch by `Darwin`-platform to automatically disable avahi and use `-mt` Version of `rpcgen` would be great 👍

lundmar, 2018-11-27:
@soundstorm Patches are welcome ;)

The proper solution would be to write the small piece of code to replace avahi with bonjour/zeroconf on mac. Unfortunately I don't have a mac myself so any support for mac would have to be contributed by others and I would welcome that very much :+1:

hraftery, 2021-03-23:
Thanks @soundstorm that worked a treat! In summary:

1. Download and unpack liblxi source.
2. Run `./configure --disable-avahi`
3. Install MacPorts and run `sudo port install rpcgen-mt`
4. Modify `src/Makefile.am`, replacing `rpcgen` on the second last line with `/opt/local/bin/rpcgen-mt`
5. `make`
6. `sudo make install`

ochococo, 2021-04-20:
**Notes from the macOS Big Sur battlefield.** 

```bash
brew install automake 
./autogen.sh 
./configure --disable-avahi
```

You need Mac ports (https://ports.macports.org) to do this step:

```bash
sudo port install rpcgen-mt   
```

After using `rpcgen-mt` in `src/Makefile.am` I was almost but not quite able to run `make`:

Got:

```c
vxi11core_clnt.c:18:2: error: implicitly declaring library function 'memset' with type 'void *(void *, int, unsigned long)' [-Werror,-Wimplicit-function-declaration]
        memset((char *)&clnt_res, 0, sizeof(clnt_res));
```

added this to `vxi11core_clnt.c`:

```c
#include <string.h>
#include <unistd.h>
```

then `make`, `make install` went fine.

Last step, `cd test`, changed to my devices ip and location, then:

```bash
gcc /usr/local/lib/liblxi.dylib scpi.c 
```

It compiled! But when I try to run it:

```bash
[1]    16707 segmentation fault  ./a.out
```

It does that on this line:

```c
 device = lxi_connect("192.168.X.X", 0, "gpib0,22", timeout, VXI11);
```

It crashes during the connection attempt is performed (I can see Agilent E5810A ethernet light flashing).

hraftery, 2021-04-20:
> **Notes from the macOS Big Sur battlefield.**

I find it surprising you experienced these issues and I did not, despite using the same OS only a month earlier with no commits to master in the meantime. Sounds like the library compilation was successful in the end. From there I compiled and ran `lxi-tools` without issue (although the GUI is buggy, but that's another matter).

I suggest you compare your `scpi.c` file with the equivalent from `lxi-tools` - if `lxi-tools` works then maybe you can find some hints there.

ochococo, 2021-04-21:
Aaaaand it works!

**Recipe that worked on Big Sur:**

Thanks @soundstorm @hraftery 🙏 

```bash
./autogen.sh 
./configure --disable-avahi
nano src/Makefile.am //  replace `rpcgen` with `/opt/local/bin/rpcgen-mt`
make
make install
cd ./test
gcc /usr/local/lib/liblxi.dylib scpi.c 
 ./a.out
KEITHLEY INSTRUMENTS INC.,MODEL 2000,0598354,A05  /A02  
```
What happened? I don't know, just tried it from scratch today and it went fine. It's not like I was able to build this thing right away, tried couple of solutions and one of the experiment must have caused this weird crash.

**Next step?** 

Build the lib for `armv7` and `arm64` :)

lundmar, 2021-12-25:
FYI - in the next release autotools is replaced with meson and the need for rpcgen is also gone.

closed 2021-12-25 by lundmar