lxi-tools/liblxi #28: Publish mail address of signing public key of liblxi at keys.openpgp.org?

robert-scheck, 2022-10-01:
Do you see any chance to publish the mail adress of the public key of RSA key 101BAC1C15B216DBE07A3EEA2BDB4A0944FA00B1, being used for signing liblxi release tarballs, at keys.openpgp.org? This would allow Fedora to switch from the current binary GnuPG output blob (manually created) to the ASCII format provided there. Unfortunately keys.openpgp.org [strips all user IDs unless the owner of the corresponding email address has allowed them to be published](https://superuser.com/questions/1485213/gpg-cant-import-key-new-key-but-contains-no-user-id-skipped), thus keys.openpgp.org can't be used fo the ASCII format right now.

lundmar, 2022-10-01:
Sure - I've verified my email with the key. You should be able to see it now.

robert-scheck, 2022-10-01:
Yes, that works now - thank you very much!

closed 2022-10-01 by robert-scheck