lxi-tools/liblxi #29: CMake Support

ztnel, 2022-10-07:
Any plan to add CMake support to this project? I would like to use this library in the future for embedded builds within a CMake project. Something like:
```cmake
FetchContent_Declare(
  liblxi
  GIT_REPOSITORY https://github.com/lxi-tools/liblxi
  GIT_TAG v2.1
)
FetchContent_MakeAvailable(liblxi)
target_link_libraries(liblxi-static)
...
```
I'm not familiar with `meson` but I'm sure a number of others would appreciate CMake support. Thanks for the library!

lundmar, 2022-10-07:
Sorry no. I do not want to use nor maintain two different build systems.

I recently converted all my open source projects from autotools to meson because autotools is starting to show its age and because I wanted to minimize maintenance. Also, meson is quickly becoming the preferred choice of new and even existing projects.

closed 2022-10-08 by lundmar

eli-schwartz, 2022-10-09:
> Any plan to add CMake support to this project? I would like to use this library in the future for embedded builds within a CMake project. Something like:

CMake's external project support should support building things that use other build systems already. You just need to override the build steps instead of the default CMake ones.

Have you tried it?