lxi-tools/liblxi #30: Refactor pthread_timedjoin_np usage?

kallisti5, 2022-10-21:
```pthread_timedjoin_np``` isn't very portable, and is a convenience function not available on stricter POSIX platforms.

Any chance of refactoring this to an alternative more standard design?

I saw https://stackoverflow.com/questions/11551188/alternative-to-pthread-timedjoin-np , seems like a common complaint.

kallisti5, 2022-10-21:
oh, lol.  The "np" in "pthread_timedjoin_np" actually stands for non-portable.

Fixed with the above PR.  It pretty much refactors the alternative to pthread timedjoin and implements similar functionality locally.  This should fix liblxi on BSD, Haiku, etc.

lundmar, 2022-10-21:
BSD is so strict :) I'll take it. Thanks.

closed 2022-10-21 by lundmar