lxi-tools/liblxi #34: lxilua.c needs a change otherwise it may get conflict with other lua "connect" functions.

roman65536, 2022-12-09:
Hi,

as from github, require "lxit" returns just true. It should return a table, so it possible differentiate between different packages.
Otherwise lua may call "connect" from socket and not from lxit. With the change below it is possible use statement as followed:

lxit = require 'lxit'

connection=lxit.connect(..)
lxit.scpi(connection,"*IDN?")
..

static const luaL_Reg lxilib[] = {
{ "connect", connect_l},
{ "disconnect", disconnect},
{ "scpi", scpi },
{ "scpi_raw", scpi_raw},
{ "sleep", sleep_},
{ "msleep", msleep},
{ "clock_new", clock_new},
{ "clock_read", clock_read},
{ "clock_reset", clock_reset},
{ "clock_free", clock_free},
 {NULL, NULL}
};

//int lua_register_lxi(lua_State *L)
int luaopen_lxit(lua_State *L)
{
/*
    lua_register(L, "connect", connect_l);
    lua_register(L, "disconnect", disconnect);
    lua_register(L, "scpi", scpi);
    lua_register(L, "scpi_raw", scpi_raw);
    lua_register(L, "sleep", sleep_);
    lua_register(L, "msleep", msleep);
    lua_register(L, "clock_new", clock_new);
    lua_register(L, "clock_read", clock_read);
    lua_register(L, "clock_reset", clock_reset);
    lua_register(L, "clock_free", clock_free);
*/

      luaL_newlib(L, lxilib);

    load_log_script(L);

    return 1;
}

regards
Roman

lundmar, 2022-12-09:
Right, I've fixed the potential naming conflict in the lxi-tools repo by prefixing all functions with "lxi_".

lundmar, 2022-12-10:
FYI - the snap edge channel contains these changes.