lxi-tools/liblxi #39: MacOS mdns (Bonjour) backend and instrument discovery

robcarruthers, 2023-05-09:
Adding to the comments at the end of #36,  I’ve now got time to spend on this task.

The [example](https://github.com/lxi-tools/liblxi/issues/36#issuecomment-1528947698) is running and printers and airplay devices respond as described.

With regards to implantation, It would seem at a high level mdns (and Avahi) is used for, obviously, instrument discovery, but also service discovery ie. when another computer/instrument publishes an LXI service the backend would catch it and call the service callback. Could this service discovery potentially be a long running background task or is it just fired up when need?

I haven’t written C code in 15 years (and I wasn’t that good at it back then), I found the The Avahi implementation is a bit contrived, could you give me a high level explanation of the backend operation and how you would like it implemented?

On first pass of the code, it would appear mdns is only used when called by the lxi_discovery functions and the discovery type is set to mdns. Is this correct or is it called in other parts of the library?

Given the initial testing, I’m reasonably confident, once I get my head around this all and with your help, getting a PR together should be fairly straight forward.

lundmar, 2023-05-09:
Great. I was hoping the sample code would just work.

Yes, mDNS is not lxi or instrument specific. mDNS is a modern way of advertising and discovering any type of network enabled device. The LXI consortium decided to make use of this feature in their newer standards.

With liblxi we typically onle fire up the mDNS for as long as we need. You decide how long via the timeout parameter.

I appreciate your effort to try make this work. And yes, Avahi can get a bit long in the tooth with all its async callbacks etc.

I think you basically need to do the following:

1. Make meson search for Bonjour mDNS dependency in case of macOS
2. Likewise, as you have a HAVE_AVAHI you need to introduce e.g. a HAVE_BONJOUR. See: https://github.com/lxi-tools/liblxi/blob/91bb485353f9e719aaa41ecf5e3be6f39e2d2177/src/mdns.c#L40
3. Same place, you simply need to call your implementation of bonjour_discover()
4. Rename your example main() to bonjour_discover()
5. Add full list of service types relevant for lxi. There are 5 types. See: https://github.com/lxi-tools/liblxi/blob/91bb485353f9e719aaa41ecf5e3be6f39e2d2177/src/avahi.c#L196
6. In your example codes resolve_callback() you then call the liblxi service callback with the following information: address, id, service, and port. The callback is passed via the lxi_info_t structure as a parameter to bonjour_discover().
7. Use the timeout parameter from bonjour_discover() in the timeval tv structure.

I may be forgetting some details but I that is the gist of it.

lundmar, 2023-05-09:
FYI - for the Bonjour meson check you may find some inspiration here:
https://github.com/postgres/postgres/blob/c5b7f67fcc8c4a01c82660eb0996a3c697fac283/meson.build#L553

lundmar, 2023-05-10:
Oh, by the way. I forgot. Regarding your question of long running service discovery. I said we only fire up mDNS for as long as we need via timeout. That is true in a certain sense. To be more precise, Avahi and Bonjour are actually running as background system services always picking up on devices reporting their presence. We simply query e.g. the Avahi daemon for it's latest discoveries and discoveries within the timeout.

This is also the exact reason a contributor failed to implement mDNS for macOS because he tried to implement it as a standalone mDNS mechanism which of course conflicted with the existing Bonjour service. Whichever system one uses, the correct way is to consult the systems mDNS service.

lundmar, 2023-05-20:
@robcarruthers #40 is wrapped up and merged - good job! 👍🏻

lundmar, 2023-05-20:
If you feel like giving a shot at creating a homebrew formula for lxi-tools that would be great. I mean, since you have now added support for liblxi on mac, it should be low hanging fruit to get lxi-tools running, at least it's command-line tool without the gui.

The gui may also be buildable. Brew has gtk4 etc. but I think I remember vaguely that one component may be missing in brew.

robcarruthers, 2023-05-20:
Agreed, the CLI is an immediate priority fro me as it is needed for several ongoing projects, but the broader objective of extending macOS support for LXI tools should come together quickly, with a brew formula to start.

For me, and I suspect many others, Liblxi stands out as an excellent solution for instrument automation, surpassing many commercial alternatives. I Appreciate your guidance, @lundmar. Your expertise truly shines through!

closed 2023-05-20 by robcarruthers

lundmar, 2023-05-20:
Thank you - I'm glad you like it.

I do my best to make a clean cut open source solution for instrument automation. I originally created lxi-tools exactly because I couldn't fathom the fact that I had to literally install gigabytes of instrument vendor bloatware just to manage my test instruments. It just seems wrong when it can be made so much simpler and still get the job done.

I'm a sw/hw engineer and I have many colleagues who need the same as I and we agree that stitching together test configuration for your DUT using various vendor components oftentimes becomes quite messy. I see you are an electronics engineer so I expect you know exactly what I'm talking about.

My goal is to try LXI enable my entire test environment. lxi-tools solved the software side of things. There is still stuff to improve like added HiSlip support and maybe even server support to liblxi. However, I'm also working on creating some open source LXI compatible HW so that I can literally do all my testing and interfacing to DUT via LXI compatible devices. Imagine how simple, clean, and flexible your test setup can become when you can do everything via liblxi/lxi-tools.