lxi-tools/liblxi #41: Avahi-client issue when installing from source on Ubuntu 23.04

robcarruthers, 2023-05-22:
I've cloned the repo and ran the meson setup which told me avahi-client and avahi-client header were not present.
I installed the libavahi-client-dev as recommended here:  https://github.com/lxi-tools/liblxi/issues/21#issuecomment-1001069828
``` shell
❯ sudo apt install libavahi-client-dev libavahi-common-dev
```
Meson setup then returned the following:
``` shell
❯ meson setup build --wipe
The Meson build system
Version: 1.1.0
Source dir: /home/rob/Code/clang/liblxi
Build dir: /home/rob/Code/clang/liblxi/build
Build type: native build
Project name: liblxi
Project version: 1.21
C compiler for the host machine: cc (gcc 12.2.0 "cc (Ubuntu 12.2.0-17ubuntu1) 12.2.0")
C linker for the host machine: cc ld.bfd 2.40
Host machine cpu family: x86_64
Host machine cpu: x86_64
Found pkg-config: /home/linuxbrew/.linuxbrew/bin/pkg-config (0.29.2)
Found CMake: /usr/bin/cmake (3.25.1)
Run-time dependency avahi-client found: NO (tried pkgconfig and cmake)
Run-time dependency libxml-2.0 found: YES 2.11.4
Run-time dependency threads found: YES
Has header "avahi-client/client.h" : YES 
Run-time dependency libtirpc found: YES 1.3.2
Configuring lxi_connect.3 using configuration
Configuring lxi_disconnect.3 using configuration
Configuring lxi_discover.3 using configuration
Configuring lxi_discover_if.3 using configuration
Configuring lxi_init.3 using configuration
Configuring lxi_receive.3 using configuration
Configuring lxi_send.3 using configuration
Build targets in project: 1

Found ninja-1.11.1 at /home/linuxbrew/.linuxbrew/bin/ninja
```
This left me a little confused as meson seemed to find the header but wasn't able to identify the library.

Upon building the test file, I was presented with numerous reference errors related to avahi functions. Here's the command I used and the result:
``` shell
❯ gcc -I/home/linuxbrew/.linuxbrew/include test/search.c -L/home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu -llxi -o search

/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `avahi_service_resolver_free'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `avahi_client_new'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `xdr_bytes'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `avahi_simple_poll_loop'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `avahi_service_browser_get_client'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `avahi_client_free'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `xdr_bool'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `avahi_address_snprint'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `xdr_int'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `avahi_strerror'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `avahi_simple_poll_quit'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `avahi_elapse_time'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `avahi_service_browser_new'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `xdr_string'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `clnt_create'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `xdr_void'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `avahi_service_resolver_get_client'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `avahi_simple_poll_get'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `xdr_enum'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `xdr_u_char'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `xdr_u_int'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `avahi_service_resolver_new'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `avahi_simple_poll_new'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `avahi_simple_poll_free'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `avahi_client_errno'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `avahi_service_browser_free'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `xdr_u_short'
/usr/bin/ld: /home/linuxbrew/.linuxbrew/lib/x86_64-linux-gnu/liblxi.so: undefined reference to `xdr_char'
collect2: error: ld returned 1 exit status
```

I suspect these error will resolve if the avahi-client is present, so whats the recommendation for installing the avahi-client correctly?

lundmar, 2023-05-23:
You are missing some key options in your compile/linker line.

If you are on a very old Ubuntu system you may be missing the avahi-client.pc file which is the feature that meson is looking for when searching for libraries automatically.

In recent versions of Ubuntu, libavahi-client-dev comes with the following file:
```
/usr/lib/x86_64-linux-gnu/pkgconfig/avahi-client.pc
```

So you can ask via pkg-config for compilation and linking flags when using avahi:
```
$ pkg-config --cflags avahi-client 
-D_REENTRANT 

$pkg-config --libs avahi-client 
-lavahi-common -lavahi-client
```

lundmar, 2023-05-23:
I just noticed you are running on Ubuntu 23.04 but I suspect you are running it the brew environment so maybe it does not find the pkg-config configuration unless you specify pkg-config as a dependency.

I'm also running Ubuntu 23.04 and in a native Ubuntu environment it finds the pkg-config avahi configuration just fine:

``` shell
lundmar@wopr:~/projects/lxi-tools/liblxi$ meson setup build --prefix=$HOME/opt/liblxi --wipe
The Meson build system
Version: 1.0.1
Source dir: /home/lundmar/projects/lxi-tools/liblxi
Build dir: /home/lundmar/projects/lxi-tools/liblxi/build
Build type: native build
Project name: liblxi
Project version: 1.21
C compiler for the host machine: ccache cc (gcc 12.2.0 "cc (Ubuntu 12.2.0-17ubuntu1) 12.2.0")
C linker for the host machine: cc ld.bfd 2.40
Host machine cpu family: x86_64
Host machine cpu: x86_64
Found pkg-config: /usr/bin/pkg-config (1.8.1)
Run-time dependency avahi-client found: YES 0.8
Run-time dependency libxml-2.0 found: YES 2.9.14
Run-time dependency threads found: YES
Has header "avahi-client/client.h" : YES 
Run-time dependency libtirpc found: YES 1.3.2
Configuring lxi_connect.3 using configuration
Configuring lxi_disconnect.3 using configuration
Configuring lxi_discover.3 using configuration
Configuring lxi_discover_if.3 using configuration
Configuring lxi_init.3 using configuration
Configuring lxi_receive.3 using configuration
Configuring lxi_send.3 using configuration
Build targets in project: 1

liblxi 1.21

  User defined options
    prefix: /home/lundmar/opt/liblxi

Found ninja-1.11.1 at /usr/bin/ninja
```

robcarruthers, 2023-05-23:
Thanks, that helps. I suspect your right as I've been going between brew initially and then from source once I encountered issues. I did manage to get the search.c file to compile without the error and get it running with this command, late last night.
```
gcc $(pkg-config --cflags avahi-client) -I/home/linuxbrew/.linuxbrew/include test/search.c $(pkg-config --libs avahi-client) -llxi -ltirpc -o search
./search
```

But I had to env variables to get working and the linker directives seem overly complicated and likely a symptom of what I've done trying fix things.

This morning I plan to, retrace my steps, come up with a better/simpler process and document here for others.

I also have a need for mdns on an 18.04 Ubuntu server, so will double check that too. Stand by...

lundmar, 2023-05-23:
If I remember correctly, the reason we also have the avahi header check, is to also work on older distros which do not include the .pc file. Though, support for old distroes is not really a priority.

lundmar, 2023-05-23:
You are trying to build search.c manually but I assume your current mission is to try create a brew lxi-tools formula?

If so, I think it will be easier if you just go straight to try building lxi-tools creating a brew formula because I expect it will be structured in much the same way as the existing liblxi brew formula. As liblxi/lxi-tools uses meson it will take care of many of the details that you have been struggling with:

https://github.com/Homebrew/homebrew-core/blob/4b4d49d83bc0d0f5c65f36356e6084e580491db8/Formula/liblxi.rb

Notice that pkg-config is a build dependency in this brew formula and that is what makes the build find the pkg-config configuration files etc.

To begin with you could just make a lxi-tools formula that does not build the lxi-gui tool by adding the meson argument ```-Dgui=false```. This would be a good start, then your formula only need to add the following dependencies: liblxi, readline, lua.

robcarruthers, 2023-05-23:
Ok, that makes sense, I'll get onto that now.

If you're happy i'll leave this issue open for now and close once we've got a better solution documented.

robcarruthers, 2023-05-24:
Ok, after some testing, the best option to get avahi running on Ubuntu 23.04 is to just install [lxi-tools](https://github.com/lxi-tools/lxi-tools#42-installation-using-snap).

```
snap install lxi-tools
```
With regards to the brew formula, mentioned above, I hope to have this submitted soon. This will enable lxi-tools on macOS also (GUI disabled). It currently works, installing from source.

closed 2023-05-24 by robcarruthers