lxi-tools/liblxi #55: How to Solve the 1024-Byte Transmission Limit

crazyRabbitboomboomboom, 2024-12-02:
Hello, I am using liblxi to communicate with the RIGOL DG1032, but I found that the maximum transmission size is only 1024 bytes. Why does this limitation exist? Is there any related restriction or rule in the source code? If I want to remove this limitation, how can I do it?

lundmar, 2024-12-02:
Normally the MTU defaults to 1500 bytes. In you case the 1024 limit may be a limitation of your test instrument. Either way, there will be a limit. Liblxi provides a low level API. If you want to send large amounts of data you will need to handle that using multiple lxi_send() calls.

For receiving there is an example here: https://github.com/lxi-tools/liblxi/blob/master/test/receive-image-data.c . You will have to do something similar for sending large data.

crazyRabbitboomboomboom, 2024-12-05:
Thank you for your answer, it was very helpful for me!

closed 2024-12-05 by crazyRabbitboomboomboom