lxi-tools/liblxi #56: lxi fails to discover device on debian base distros

emi-mpspace, 2025-01-09:
I tried lxi and lxi-tools on several debian (debian, ubuntu 24.04) distributions and i'm failing to connect to the an lxi device. 

Not sure if related, but I tried compiling the sample code from the README file and it does not compile under debian based distros due to linking errors (i do have all the dependencies and specified the library path).

The command does work when installing through snap, but my intention is to run lxi inside a docker container where snap is not an option.

emi-mpspace, 2025-01-09:
Forgot to mention that it does work on redhat distros (at least on fedora)

lundmar, 2025-01-09:
Please provide more details.

psalic, 2025-01-12:
lxi tools can't compile with debian bookworm because gtk4 library in bookworm is old. 
lxi-tools 2.8 was added in debian trixie which is still in  testing phase.
Even if trixie is testing you can run trixie in container and lxi tools should work.